OXRS-IO/OXRS-IO-TouchPanel-ESP32-FW #16: Add ability to set tile backgroud colour

MakerDockio, 2022-08-21:
It would be great if we could have the facility to specify a colour for a tile, something like this.

```json
{
  "tiles":[
    {
      "screen":<number>,
      "tile":<number>,
      "backgroundColorRgb":{
        "r": <number>,
        "g": <number>,
        "b": <number>
      }
    }
  ]
} 

```

Given the naming convention here, it would make sense to change 'Set icon color' to:

{
  "tiles":[
    {
      "screen":<number>,
      "tile":<number>,
      "iconColorRgb":{
        "r": <number>,
        "g": <number>,
        "b": <number>
      }
    }
  ]
} 

```

this just makes the payloads a little more self explanatory.

moinmoin-sh, 2022-08-30:
The colored tle background would only be visible in OFF state because the ON state is defined as white. Is it that what you want to achieve? If not, please describe what you are looking for.

MakerDockio, 2022-08-30:
Yes I think that's ok.

PR #29 (2022-09-03): add option to set individual tile background

```json
{
  "tiles":[
    {
      "screen":<number>,
      "tile":<number>,
      "backgroundColorRgb":{
        "r": <number>,
        "g": <number>,
        "b": <number>
      }
    }
  ]
} 

```

closes #16
closed 2022-09-03 by moinmoin-sh