OXRS-IO/OXRS-IO-TouchPanel-ESP32-FW #23: set tile to offline / online

austinscreations, 2022-08-31:
I've been working on rebuilding @sumnerboy12 led controller sub flow to allow more control from multiple interfaces. something that both HA and NR have is the ability to set an element to an offline state so you can't click or activate it. think this may be potentially useful on the screen. 

my other thought i just to use teh sub label to display offline state, and ignore the messages but wonder if something better would feel nicer overall.

moinmoin-sh, 2022-09-02:
I suggest we "grey out" the tile and disable touch sensitivity so no clicks are generated. The tile would still be able to show all state information.

moinmoin-sh, 2022-09-04:
proposal with tile 3(on state)  and tile 4 (off state) disabled

[image]

sumnerboy12, 2022-09-04:
That looks good to me Moin. Pretty clear and obvious.

austinscreations, 2022-09-04:
looks / sounds good to me

PR #31 (2022-09-04): add option to disable a tile
Add an option to disable a tile.
A disabled tile doesn't accept touch events and the tile is greyed out.
cmnd/ payload  `{"tiles":[{"screen":"1", "tile":"3","disable":true}]}`
[image]

closes #23
closed 2022-09-04 by moinmoin-sh