OXRS-IO/OXRS-IO-TouchPanel-ESP32-FW #25: Ability to display plain text or HTML on a screen

hazymat, 2022-08-31:
Rather than tiles / keypads etc., would be useful to be able to display plain text or even HTML on a screen.

Useful for serving latest news / weather...

sumnerboy12, 2022-08-31:
I was thinking this would be a new tile style, which shows a label/icon/summary on the tile and when clicked launches a fullscreen popup where you can fill it with text/HTML/markup.

hazymat, 2022-08-31:
That would be even better than what I said!

moinmoin-sh, 2022-09-02:
Rendering anything else than plain text is an issue here. LVGL doesn't offer any format changes within a label widget (basically a text box). The only formating options are "\n" for a line break and the recoloring as we use it for the textIcon and that doesn't work across line breaks.

caitken-com, 2022-09-05:
Even without what @moinmoin-sh just said about unable to format text, remembering that html requires a massive parser. Doubt you could get one to work on an ESP.
Best bet would have been a sub-set of Mark-Down.

moinmoin-sh, 2022-09-05:
Given the constaints we have there will be no chance to parse and display any standard mark-up text. To show anything other than plain text a FW specific formating has to done by the backend.
As an alternative to plain text the information could be presented in a "feed style" like pop-up.
Each "feed" has a `heading `and a `body`. Character based coloring is possible.
Could look like this
[image]

sumnerboy12, 2022-09-05:
That looks pretty cool I think Moin, should be able to do plenty with those tools.

PR #38 (2022-09-11): add TS_FEED tile style to show message feed
EDIT:  feeds are tile dependent 

standard :
  chevron-right  `>`  chosen as top-right indicator for tile type `feed `
  new posts are pushed to front (shown as top post on feed)
  holds up to 5 posts, if > 5 -> oldest post removed

 
elements :
`id` : optional, can be any number, needed if id-related deletion is wanted
`head `: optional head of post  
`body `: optional body of post
at least one of `body `|| `head `is required, otherwise ignored

proposed payloads :  ( I assume you have better ideas for the payloads )

add post to the top of the list [... cut]
closed 2022-09-11 by moinmoin-sh