OXRS-IO/OXRS-IO-TouchPanel-ESP32-FW #33: Remove requirement to have first screen

hazymat, 2022-09-05:
During my initial playing around with the firmware I dynamically configured some screens (ie wrote a NodeRed flow that could automatically add a screen with tiles depending on some other external condition). When removing the first screen ie Screen ID #1, my hope was that the next screen ID would be shown first.

(As an aside, if you upload custom icons to screen ID #1 then remove that screen, the screen plus its tiles with no icons is shown still, which seems janky.)

It’s doubtless always possible to workaround this externally, ie shift all the screens down by one when you want to effectively delete screen 1, but arguably it just makes it a bit more fiddly to work with.

Totally understand if this is not priority, but how easy is it to remove this limitation?

moinmoin-sh, 2022-09-11:
I compare the screen1 (Home Screen) to the home screen of a smartphone. It's kind of a constant in this very flexible system.
In other words, I would like to keep this "restriction".

But your issue highlights something else (which might be related to your issue) : 
We don't have an API to clearly delete tiles and screens. This would be essential for a backend as you introduced which dynamically creates/deletes screens and tiles.

hazymat, 2022-09-11:
Thanks for separating those two concepts out - agree that it would be beneficial to delete tiles and screens. This was one of the first thing I struggled with. Instead I just re-upload screens then reboot. This works to "remove" screens from before. Problem is that if you don't replace the first screen with something then those tiles persist somehow, even though they are not configured to exist.

So working on the basis that there should be an API for deleting tiles and screens, then deleting tiles from the first screen would work, but deleting the first screen would essentially do nothing. That would work from where I'm sitting!

moinmoin-sh, 2022-09-11:
I assume these persisting tiles were configured via the adminUI which stores the configuration in SPIFFS and restores on start up.
You could use the adminUI to delete all configured tiles or erase the flash, program via USB and start from scratch.

PR #42 (2022-09-21): add cmnd/ payloads for screen and tile removal
- tile "disable" has moved to "action":"disable"|"enable"

new cmnd/ payloads to remove screens and tiles

**remove screen:**
```
{
  "screens": [
    { "screen": 2, "action": "remove" }
  ]
}
```
**remove tile:**
```
{
  "tiles": [
    { "screen": 1, "tile": 2, "action": "remove" }
  ]
}
```
braking change : keyword "disable" for tiles removed
new "action"
**disable / enable tile:**
```
{
  "tiles": [
    { "screen": 1, "tile": 2, "action": "disable" | "enable" }
  ]
}
```

rules:
- removing a screen removes all tiles on this screen as well
- rem [... cut]
closed 2022-09-21 by moinmoin-sh