OXRS-IO/OXRS-IO-TouchPanel-ESP32-FW #35: bug: in disable tile

austinscreations, 2022-09-07:
bug in #31 
running Ethernet v1.6.1
sending `{"tiles": [{"screen":1,"tile":1,"disable":1}]}` more than once causes the tile to get continuously darker. but your only able to send `{"tiles": [{"screen":1,"tile":1,"disable":0}]}`  once to bring the tile back to normal state - once doesn't work if your flow accidentally sends the payload to disable more than once.

MakerDockio, 2022-09-07:
Just to confirm isn't the disable property value supposed to be a boolean?

cmnd/ payload 
{"tiles":[{"screen":"1", "tile":"3","disable":true}]}

cmnd/ payload 
{"tiles":[{"screen":"1", "tile":"3","disable":false}]}

austinscreations, 2022-09-08:
Not as far as I could tell in code, and I wasn't getting a response for boolean `false` but it would act on `true`
Same issue still persisted though if using boolean

sumnerboy12, 2022-09-08:
I think this is a valid bug - when sending `"disable":true` repeatedly it doesn't clear the existing foreground tile which provides the disabled/dimming effect. So it just keeps adding more of these foreground tile covers over the top of one another.

commit e8ea301f (2022-09-08): fixes #35
commit 048fc590 (2022-09-08): fixes #35 where disabling a tile multiple times creates multiple overlaps (#37)
closed 2022-09-08 by sumnerboy12