OXRS-IO/OXRS-IO-TouchPanel-ESP32-FW #47: Enhancement request - Screen sleep timeout - Turn down option, rather than Turn off

dweston, 2022-10-24:
Currently the `noActivitySecondsToSleep` parameter sets a timeout till the screen backlight is 'Turned off'.

Is it possible to enhance this function such that the screen backlight can be turned down, rather than only turned off. In dark, no light, situations it would be a huge boon to the utility of these screens to be able reduce the backlight illumination to a very low level - in the region 1 - 5% - in order that users can locate a screen in total darkness?

I acknowledge that illumination even at this low level could mean screen life extension is significantly compromised versus completely switching the screen off. However, there are other methods for contending with this, such as area proximity sensors in entrances or hallways that can be used to 'wake' a number of screens from complete blackness within in a designated area, if potential users enter that region. And then, once the area is 'empty' again, the screens can all be turned off.

sumnerboy12, 2022-10-24:
See https://oxrs.io/docs/firmware/touch-panel-esp32.html#control-the-backlight.

You can set the backlight level via an MQTT command - I am using this for example, to constantly update the screen backlight level depending on the time of day.

dweston, 2022-10-25:
Yep - I’m across that and have been using that command to recreate the`noActivitySecondsToSleep` timeout functionality in node red, which has turned out to be pretty involved.

Ultimately I’m suppose I’m a self interested advocate of seeing that all the OXRS ecosystem kit is sufficiently ‘clever’ to cover most of what I’d characterise as basic hygiene functions, like product life extension and energy efficiency tasks, in firmware, cognisant, of course, that RAM on these devices is extremely limited leading to the "thin client” ethos.

The other driver is striving to reduce the complexity overheads for new adopters to the degree that sensibly can be achieved, so ‘out of the box’ the user is focussing on and receiving core utility benefit with the least overhead in depth of knowledge.

If it's a bridge too far to add ’Turn down’ to the ’Turn off’ device level functionality on ’sleep’, so be it.

sumnerboy12, 2022-10-25:
This firmware is not really designed to work that way. It is more of a set of tools, which can be used with an external rules engine or server backend, to perform the clever stuff. So very little "smarts" or logic is built in, by design. Hope that makes sense.

dweston, 2022-10-25:
Indeed I well understand and support that “thin client” ethos. 

Its just that the config options do currently allow for a variability in `time till sleep`, so I figured that adding one additional config variable defining the `sleep backlight level` would not impose too onerous an overhead, given that the code to actually implement the backlight level change on sleep timeout is already there. Yes it does involve declaring an extra settings variable, and a few extra lines of code to extract and update that additional variable from the config mqtt payload. I think the following suggestions would do it:

"uint32_t _noActivitySleepScreenLevel  = 0L;
"
and

“  if (json.containsKey("noActivitySleepScreenLevel"))
  {
    _noActivitySleepScreenLevel = json["noActivitySleepScreenLevel"].as<int>();
  }
"
Which, from the extent my lay comprehension of these things goes, would seem to be the extent of it, save for the change required to add the new variable in place of the ‘0’ in the function call `_setBackLight()` in this section:

“  if (_noActivityTimeOutToSleep != 0)
  {
    // is in sleep allready ?
    if (_actBackLight > 0)
    {
      // time elapsed, jump to HomeScreen
      if (lv_disp_get_inactive_time(NULL) > _noActivityTimeOutToSleep)
      {
        _retainedBackLight = _actBackLight;
        _setBackLight(_noActivitySleepScreenLevel);
      }
    }
  }
"
Forgive my ignorance if I’ve missed something here. I don’t have the dev environment set-up to suggest these changes in GitHub.

sumnerboy12, 2022-10-26:
You raise a valid point - I will let @moinmoin-sh have a look and comment.

moinmoin-sh, 2022-10-26:
The backlight now has 2 clear operational states :
0 -> sleep (only set when timed out to sleep or set to sleep via cmnd/)
1 .. 100 -> awake (set by the slider or via cmnd/)
This results in  a retained backlight always in the range 1.. 100. (higher than sleep value)

When the sleep level would be adjustable things ar less distinct. Lets say the sleep level was set to 10 and the user set the actual backlight to 5. Now timeToSleep elapsed and the level would be set to 10 (so higher than it was set by the user). This could be intended behavior but also couldn't. To avoid such situations I would prefer to stay with the simple but easily predictable FW behavior and leave everything more advanced to the user (automation).
Sumnerboy has already explained our design goals.

I definately see your point but not sure how smart the TP schould be. We can leave this issue open and see if others jump in and vote for having this feature built in.

moinmoin-sh, 2023-09-02:
No activity for 10 months. I close it for now.

closed 2023-09-02 by moinmoin-sh