OXRS-IO/OXRS-IO-TouchPanel-ESP32-FW #50: Support WT32-sc01 plus

PlPt, 2023-02-05:
hey,

is it possible to support wt32-sc01 plus board with this firmware?
Board WT32-sc01 plus has a TFT Display without an SPI, so I think it will be difficult to adapt. Somebody has tried yet?

PlPt, 2023-02-26:
I tried it out and yes, it is possible, biut the SC01 plus has a MCU8080 interface, so the TFT_eSPI lib does not support the display driver.
What to do:

1. Switch to https://github.com/lovyan03/LovyanGFX as DisplayDriver
2. Implement a Driver Class for SC01Plus like this https://gist.github.com/sukesh-ak/610508bc84779a26efdcf969bf51a2d1
3. Change flush and readTouch Methods in Main, that it uses the lovyanGFX Driver

Then it works for almost any display, supported by LovyanGFX (which are more than TFT_eSPI)
I have a quick and dirty solution wthich works, maybe you're interested in impleenting it, so more devices are supported.

moinmoin-sh, 2023-03-04:
An updated version of this FW that will support the WT32-SC01-plus is under progress and will be released soon.
Stay tuned.

moinmoin-sh, 2023-03-09:
solved with FW 2.0.0  #51

closed 2023-03-09 by moinmoin-sh