OXRS-IO/OXRS-IO-TouchPanel-ESP32-FW #52: Show intended platform in FW details

moinmoin-sh, 2023-03-15:
I think we should add a row in the FW-details of the `Settings `screen (and the json "frimware") that tells what the loaded FW was built for.
like 

`Platform : WT32-SC01`
or
`Platform : WT32-SC01-PLUS`

Not sure if `Platform` is the right key?

moinmoin-sh, 2023-03-16:
@sumnerboy12 any thoughts on this?

sumnerboy12, 2023-03-16:
Ha - I was literally just looking at this and thinking about my response. We already have `ESP32` in the firmware name - would it make more sense to change this for each version of the firmware we build - i.e. `OXRS-IO-TouchPanel-WT32SC01-FW` and `OXRS-IO-TouchPanel-WT32SC01PLUS-FW`? This would mess up the release binary filenames a bit tho...

moinmoin-sh, 2023-03-16:
... and the name would become a little too long to be displayed on the screen

plains203, 2023-03-16:
I think Moin's idea would be neater there. I would try and keep it out of the firmware name because the name reflects the features.

sumnerboy12, 2023-03-16:
yep think you are right - the name shouldn't change between versions - so just the JSON key to decide on - `platform` or `hardware` or what?

moinmoin-sh, 2023-03-16:
I think `platform `is pointing more into the `MCU `direction `ESP32`. `hardware `seems to be more suitable to me. 
Or may be `device `?

sumnerboy12, 2023-03-16:
Hardware seems best, it is the "hardware" the "firmware" is supposed to be run on.

plains203, 2023-03-16:
I like hardware

moinmoin-sh, 2023-03-16:
agreed !

kghxkgh, 2023-03-29:
Now that you agreed, can I throw in the term "Build", I was just thinking Hardware might be confusing with the actual hardware it is running on.

closed 2023-04-16 by moinmoin-sh