OXRS-IO/OXRS-IO-TouchPanel-ESP32-FW #53: Implement the level UX from the new Web-TouchPanel

sumnerboy12, 2023-04-01:
Instead of the little popup in the top-right of the tile to show the level bar as you change it from the device, make the entire tile a bar whose level changes depending on state.

moinmoin-sh, 2023-04-02:
Could you live with rounded corners of the bar?
TBH I'm not so happy with this layout. Looks cluttered to my because icons and text are somehow broken when the bar is dividing them.
No clue how to handle the label text to make sure it's hown properly when divided.

Do you have the APP in production (real life examples)?

[image]

sumnerboy12, 2023-04-02:
[image]

sumnerboy12, 2023-04-02:
[image]

sumnerboy12, 2023-04-02:
I think your solution looks great

moinmoin-sh, 2023-04-03:
Not a solution yet - just a POC to understand what would be possible.

sumnerboy12, 2023-04-14:
Think this can be closed - what you have done looks great. Just need a release with it included :).

closed 2023-04-14 by sumnerboy12