OXRS-IO/OXRS-IO-TouchPanel-ESP32-FW #69: Label cannot be changed via cmnd topic

smcgann99, 2023-09-29:
I just noticed that I can change a lot of properties on a button via cmnd, but not label, this seems a bit inconsistent.
It also makes it difficult to change the tile dynamically as we have to send some changes to **cmnd** and some to **conf** topics.

I cannot think of a good reason for this, so maybe its an oversight ?

moinmoin-sh, 2023-09-29:
I see the `label` as a static (only configurable) item of the tile. The `subLabel`in turn was meant to add dynamic information to the label portion of the tile.

If there is a common interest to make the `label `changeable it can be added to the `tile `command.

Do you have use cases for changeable labels?

sumnerboy12, 2023-09-29:
I think in the interests of consistency it makes sense to allow the tile label to be updated via commands.

plains203, 2023-09-29:
I could make use of this being an updatable field. It would seem to keep with the feel of the firmware to do that as well.

smcgann99, 2023-09-29:
I found the restriction when trying to "abuse" a button tile to be a weather tile ;-) 
The label could be used to display some dynamic text in different colours. 

As the audience for OXRS grows there are likely to be many novel uses, I don't think allowing this, is in any way detrimental and just adds to the possible innovations.

moinmoin-sh, 2023-09-29:
I think we have an agreement.

The requested feature will we integrated shortly.

commit b31d53db (2023-10-01): add ability to set tile label via cmnd/

closes #69
closed 2023-10-01 by moinmoin-sh