OXRS-IO/OXRS-IO-TouchPanel-ESP32-FW #70: Feature Request: Some styling options to be configurable in settings

smcgann99, 2023-09-29:
While I love the design principles and overall look, everyone likes to tinker, just think how many apps have themes ;-) 

For some such as me, who are **"hard of seeing"** some elements on the screen are a bit of a problem.
So personally I would prefer that the tiles stand out a bit more from the background when off, and be a bit less bright when on.
Also I find the faint subLabel very hard to read.

I asked on discord about **A way to set the tile default "on" background colour** but it seems there isn't one.
I had a look in the code and found the opacity settings, and thought that just being able to tweak those in the setup screen would most likely be enough, to adjust the contrast.

Obviously being able to set default font sizes for each element would also be a great help to the **visually challenged**.
Perhaps you might address these options in a future release ?
Thanks

PS not a criticism !!

moinmoin-sh, 2023-09-29:
Regarding tile contrast:
The contrast settings (opacity) as we have now were a recommendation from our UI expert. It turned out to be nicely visible on "real" screens (PC/tablett) and on the smaller (320x480) TouchPanel devices. The newer (480x480) panels, which are not widely used so far, have obviously a different brightness curve. They appear darker and there is less contrast at low intensity levels. It becomes better if a brighter background is set like a dark grey or blue. You have one of those new panels and so you see the above effect.

I could make the opacity settings for the tiles  (on and off) configurable as system wide config parameter. It needs a little caution and make sure there are no side effects but should be doable.

Regarding font sizes:
This a different story. Could you please move this topic to a seperate issue. Thanks.

smcgann99, 2023-09-29:
Thanks for considering this!
I already adjusted to a dark blue and it does help some. If you do add settings, could this also include the text / subtext please ;-)

commit 8ec2d555 (2023-10-01): add ability to configure tile brightness for ON and OFF state

closes #70
closed 2023-10-01 by moinmoin-sh