OXRS-IO/OXRS-IO-TouchPanel-ESP32-FW #71: Feature Request: Font size to be configurable in settings

smcgann99, 2023-09-29:
From a usability point of view it would be advantageous to have an option to alter the default font size used on the tiles.

I note that long text lines wrap upwards towards the icon, but even a quite big increase in size should still allow a more readable label, even if it splits over two lines. It would hold less characters, but should still be enough for most needs, and could be adjusted to suit each user.

I guess if the icon size was also adjustable then users could strike a balance between what works best for them. 
Bigger text smaller icon or vice versa.

moinmoin-sh, 2023-10-01:
The real estate of a single tile is limitted. Further constraints are caused by the used LVGL and by the design concept. 
The layout of screens and tiles has been developed and optimized over several iterations and is somewhat hardcoded. 
Making the layout configurable while keeping a nice look is unfortunately not as simple as it sounds.
Given the high flexiblity in using the TouchPanel FW we tried to limit the configurable options to a well aranged set. It wasn't a design goal to make everything configurable that could. 

As a current outlook: such configuration options will not be available in near future.

smcgann99, 2023-10-03:
Fair enough, I will see if I can get some better glasses then ;-)

moinmoin-sh, 2023-10-07:
Not planed to support.

closed 2023-10-07 by moinmoin-sh