OXRS-IO/OXRS-IO-TouchPanel-ESP32-FW #78: Allow screen load event to start / extend noActivitySecondsTo[Home|Sleep|Lock]

hazymat, 2023-10-08:
In line 2446 of main.cpp there is the function `void checkNoActivity(void)`

Currently this checks for inactivity on the display.

When you send a screen load event over MQTT:
````json
{
    "screen": {
        "load": 2
    }
}
````
this does not trigger or extend this activity. This means the screen load event doesn't work, if the activity timeout has passed.

In certain use cases, this is no problem. For example, if the screen load event is sent because I touched the screen.

However in some use cases this can be annoying, for example, if I want the screen to come change on some external event e.g. I walk into another room (presence sensor or something like that).

I think a screen load event should really "force" that screen to load, also intuitively I think it should actually make the screen light up and start that timer again, just the same as if you tap the screen.

plains203, 2023-10-08:
I like that it doesn't light the screen up when you programmatically change the screen via the backend. You can send a second command that lights it up. There are conditions where I want the screen to change to a different screen based on an event but I don't want the screen to light up and until touched. Time of day change as an example.

hazymat, 2023-10-08:
> There are conditions where I want the screen to change to a different screen based on an event but I don't want the screen to light up and until touched.

I had considered this and didn't want to add additional complexity to my request, but I do agree and think it would be a very useful enhancement.

In your example, when the screen is changed due to time of day, I presume you have set the `"noActivityToHome": 0`, to prevent the device returning to home screen? If so there are various ways of approaching this logically:

1) **Option 1** - a bit hacky - is to make the screen load event trigger and extend the timeout, but check for `"noActivityToHome": 0` and prevent the screen backlight actually coming on in this case. (Maybe there is some additional stuff that needs doing, but this is the general idea.)

2) **Option 2** - not so elegant - is to accept a new parameter with a screen load event e.g.
````json
{
    "screen": {
        "load": {
                "number": 2,
                "activateBackLight": false
        }
    }
}
````
This is not so elegant, because there's an additional requirement for the user to remember to set `"noActivityToHome":0`, otherwise it will just revert to home. (Unless there's another exception in software that says "if screen load event was sent, then on this occasion, cancel the `noActivityToHome` timer, but not for future events" - even more complexity.)

3) **Option 3** - annoying for the user whose device has a `noActivityToHome` setting - is to accept a second command to activate the display like a "phantom touch" - basically bring device out of sleep. I don't like this option because it means that, in order to actually change the screen (when timeout has passed) you need to send two messages. i.e. a screen load event won't actually load a screen on its own except if (a) the timeout hasn't passed or (b) you send a second message.

4) **Options 4** - there was another option in my mind but I've forgotten it, I'll come back here and edit the comment when I remember!

Very interested to hear the thoughts of @sumnerboy12 as he's excellent at solving these problems, where you need to keep things simple but powerful.

plains203, 2023-10-08:
I do indeed have "noActivityToHome": 0 set and have the backlight turn off after 30 seconds. 

I'd also be keen to hear some other ideas in around how best to handle it. 

I think the non breaking change is that you just send a wake-up command when you want it to and I believe that also starts the timer again? The timer is really only used for the return to home and backlight timeout? I would guess that the return to home won't then trigger until you wake the screen and then let it sit idle. 

You did mention though that the timer doesn't reset on screen load which if the screen is on and counting down leads to unexpected behaviour. Perhaps the answer is to trigger the timer and leave the backlight alone on screen change commands.

hazymat, 2023-10-08:
> The timer is really only used for the return to home and backlight timeout?

There's also a lock timer which I think is related to keypad lock, but haven't used / tested that yet.

> the timer doesn't reset on screen load which if the screen is on and counting down leads to unexpected behaviour

Yes, so there could be the case where the timer has 100ms left before the screen goes blank / returns home, and you send a screen load event expecting to see that screen, but it immediately goes blank.

Your idea to trigger the timer on screen load (but do not change backlight) is the best. I guess this would require a new variable though, because currently when you trigger the timer, the backlight comes on. I've glanced at the code but don't fully understand how it works, hopefully @moinmoin-sh can give his thoughts too.

moinmoin-sh, 2023-10-08:
Let me summarize what I understood and propose:

Requirement:
- the panel shall become on (backlight on) and show a selected screen depending on something that the backend (NR) has detected

Comments/proposal:
- I don't like the idea to add any "under the hood" behavior like connecting the "screen on" to a panel load event
- cleanest solution is to add a new `cmnd/` that awakes the panel in the same way as if a touch was detected. This would do the same things as a touch and ensures everything in the FW stays in sync.
- loading the desired screen after awaking the panel ensures the desired screen is shown
- if there was a keypad activated in the meantime you have to close it in favour of showing your selected screen
- I see value in having the "awake" `cmnd/`

I hope the above would help to solve your use case

hazymat, 2023-10-08:
Thanks for the clear summary @moinmoin-sh.

What about the problem that when you send a screen load event, if the screen is asleep then it doesn’t load that screen? It seems a bit strange. Maybe it is not a problem at all, but my expectation. Perhaps it can be ignored, but we would need to ensure the user knows they must send the awake command **before** the screen load command.

moinmoin-sh, 2023-10-08:
I tried loading a screen with the panel in sleep and it worked - don't see why it shouldn't. Not sure if there are edge cases? 
My summary was meant to describe the safe process.

sumnerboy12, 2023-10-08:
Sounds like this has been sorted, logically at least?

moinmoin-sh, 2023-10-09:
Good news: we have that already!

I was a little sidetracked yesterday. 
There is the `backlight` cmnd/ already which does exactly what I decribed.
Just send  
```json
{
  "backlight":
    {"state":"awake"}
}
```
And with the `load screen` in the **same payload** he sequence doesn't matter
```json
{
  "screen":
    {"load": 2},
  "backlight":
    {"state":"awake"}
}
```

plains203, 2023-10-09:
Correct I use that already. Not sure though about the timer issue that Matt was mentioning because I haven't checked that behaviour.

moinmoin-sh, 2023-10-09:
There aren't real timers. The timeout activities (backlight, homescreen, keypad) are looking at `time expired since last touch` and become active when their configured timeout was reached. The `time expired` is reset on every touch.
So when ` "state":"awake"` is sent, the touch is simulated and all timeouts are kind of "restarted" synchronously.
For safe operation the described sequence has to be followed or put into a single payload (see above).
I hope that makes sense.

hazymat, 2023-10-12:
Thanks @moinmoin-sh. I will update the docs according to this safe operation sequence.

I will close this issue now.