OXRS-IO/OXRS-IO-TouchPanel-ESP32-FW #88: Local i2c sensor

nofoxgit, 2024-01-24:
I can see something like "SHT2x not detected" when my WT32-sc01 starts. Can we connect local i2c sensors to touch panel ?? I can't find anything about it in the docs.. I want to connect SHT31 and send the temp and hum values over mqtt to home assistant, it would be a great feature.

moinmoin-sh, 2024-01-24:
The FW has builtin support for a SHT20 sensor which is part of the WT32S3-86S panel (480x480). This support is always activated when a SHT20 sensor is detected on the I2C bus (all supported panels). The WT32-SC01 has the I2C bus pins broken out on the expension headers. You can connect a SHT20 to these pins and the support in FW will be enabled automatically. As a result temperature and humidity are published by the tele/ topic in intervalls of 60 seconds. You'll see the "climate" reported on the "config" screen as well.
Don't know if the SHT31 would work as well.

moinmoin-sh, 2024-01-27:
Yes sorry, the docs are a little behind ...

closed 2024-01-27 by moinmoin-sh