OXRS-IO/OXRS-IO-TouchPanel-ESP32-FW #91: Include brightness in colorPicker state payloads for RGB events

sumnerboy12, 2025-05-01:
Currently we only include a `brightness` value in the state payload for `colorPicker` events, if "colour temperature" is being adjusted. 

For RGB events we return the RGB value with the brightness applied to the RGB value - i.e. if you pick red only you have RGB = 255, 0, 0. Then if you change the brightness of red to 50% you get RGB = 128, 0, 0.

This makes sense but it does not mirror how Home Assistant works. That expects the raw RGB value and then a brightness value as well.

This is also how the widget displays the value, you have an RGB colour wheel and a brightness slider.

I think it makes more sense to return the raw RGB value + brightness, and let any downstream system apply the calculations if required.

I realise this would be a breaking change but I think the more we make this stuff HA compatible the better.

moinmoin-sh, 2025-05-01:
I'm a little estonished about the HA definition for RGB colors to seperate color and brightness.

Could you please point me to the relevant HA documentation.

sumnerboy12, 2025-05-01:
It isn't mentioned explicitly but they talk about how MQTT based lights are
controlled here... https://www.home-assistant.io/integrations/light.mqtt

You can see where they talk about HS based color control, there is no V.
They are just expecting the raw color along with an explicit brightness.

I can confirm this is what HA is sending when i adjust the color and
brightness of an RGB light thru the HA dashboard.

moinmoin-sh, 2025-05-01:
I understand that you want to make the panel more Home Assisant compatible.

If we make a breaking change and use the IMO unusual split between RGB and Brightness anyway, we might use the `HS color mode` from HA instead.

[https://www.home-assistant.io/integrations/light.mqtt#hs-color](https://www.home-assistant.io/integrations/light.mqtt#hs-color)

This change would allow a much cleaner data handling in our code (no need to convert RGB <=> HSV, brightness is part of the variable)

sumnerboy12, 2025-05-01:
I think if we output HS values from the tile events, along with RGB, it could certainly help for those bulbs that support `hs` color modes. Allowing direct conversion from tile events to HA commands.

In my use-case currently the bulb supports `color_temp` and `rgb` colour modes, along with a separate brightness value. So I still need the RGB values, separately from the brightness. 

I recently updated the LED PWM firmware I wrote for Austin to be HA compatible. See https://github.com/sumnerboy12/OXRS-BJ-LightStrip-ESP-FW. In that I expect raw RGB values in the range 0-255 along with a brightness value, which I apply to the RGB values on-device, to work out the final PWM values for each LED.

This seems to be the way most devices work - they take raw RGB + brightness values and calculate the individual channel values on-device. Rather than expecting the RGB values to be pre-calculated.

This also means the device can report back the brightness of the device in status messages, which it cannot do if it receives pre-calculated RGB values (with brightness already applied).

moinmoin-sh, 2025-05-02:
OK, agreed. I see that you have the better experience.

Let's go with your PR as you prepared.

commit a666a9c2 (2025-05-02): fixes #91 (#92)

* fixes #91

* simplify brightness control and sync color/white brightness sliders
closed 2025-05-02 by sumnerboy12